riscv/riscv-isa-manual #1392: Do the following PMA changes of implicit PTE access need xFENCE to order?

terranfund, 2024-05-10:
First of all, I want to ensure PMA of PTE access is permitted to cache them the address translation cache.
If the above answer is true, I want to ask the following scenarios whether xFENCE is required to synchronize PMA setting and virtual memory system.

1.  The location of PTE is changed
2.  The cache-ability of PTE is changed
3.  The memory attributes of PTE is changed from memory region to IO/vacant region, or vice-versa.

gfavor, 2024-05-11:
On Fri, May 10, 2024 at 2:26 AM terranfund ***@***.***> wrote:

> First of all, I want to ensure PMA of PTE access is permitted to cache
> them the address translation cache.
> If the above answer is true, I want to ask the following scenarios whether
> xFENCE is required to synchronize PMA setting and virtual memory system.
>
>    1. The location of PTE is changed
>    2. The cache-ability of PTE is changed
>    3. The memory attributes of PTE is changed from memory region to
>    IO/vacant region, or vice-versa.
>
> Since PMA details are very implementation-specific (and not specified by
the architecture), and PMAs are managed by M-mode firmware, it is up to
that implementation-specific firmware to do any necessary sfence'ing when
PMAs are changed.  This is true for all memory accesses, not just PTE
memory accesses.

allenjbaum, 2024-05-11:
Configurable PMAs are custom and implementation defined, not defined by the
spec, so I suspect there is a "correct" answer.

terranfund, 2024-05-11:
OK, got it. Thanks a lot.

closed 2025-12-09 by aswaterman