riscv/riscv-isa-manual #1393: Question: How should I handle pending software interrupts delegated to Supervisor when mstatus.SIE is not enabled ?

iamkarthikbk, 2024-05-10:
On a machine with supervisor support, what happens to pending interrupts that are delegated to the Supervisor when `mstatus.SIE` is not set? Are they simply not taken? Or should the delegation not happen because SIE is not set?

for example, if I do the following in machine mode:
1. enable interrupt delegation for supervisor software interrupts `mideleg = 0x2`
2. mark a software interrupt to be pending `mip = 0x2`
3. set the machine's target privilege to Supervisor, and target interrupt enable `mstatus = 0x880`
4. perform an `mret` followed by a `wfi`, and attempt to fetch an arithmetic instruction
Should my hart take a trap with an `scause` value `0x80000002` and jump to `stvec`?

aswaterman, 2024-05-10:
The global interrupt enables, like `mstatus`.SIE, have no interaction with delegation.  Delegation occurs anyway.  Since interrupts are globally disabled for S-mode, any interrupts delegated to S-mode simply won't be taken while in S-mode.

In your example, the WFI instruction will retire and the hart will continue executing after the WFI (because an unmasked interrupt is pending), but since S-mode's global interrupt enable is off, the hart will not jump to `stvec`.

iamkarthikbk, 2024-05-13:
Thanks, that answers my question.

closed 2024-05-13 by iamkarthikbk