riscv/riscv-isa-manual #1394: I downloaded `RISC-V ISA Manual, version 20240411` and some of these extensions have a star (*), e.g. `*Zbkb`. I don't see a footnote or something to explain it. Is this a typo and this was suppose to be bold?

aswaterman, 2024-05-10:
I downloaded `RISC-V ISA Manual, version 20240411` and some of these extensions have a star (*), e.g. `*Zbkb`. I don't see a footnote or something to explain it. Is this a typo and this was suppose to be bold?

The table suggests that `Zbkb` is ratified, but looking at `28.4.5. Zbkb: Bit-manipulation for Cryptography` I see a not `The Zbkb extension is frozen`. Looking at wiki ( https://wiki.riscv.org/display/HOME/Ratified+Extensions ) I see that `Zbkb`  was ratified in `November 2021`.

The same applies to `Zbkc`, `Zbkx`. I didn't check the rest.

_Originally posted by @davidlt in https://github.com/riscv/riscv-isa-manual/issues/1299#issuecomment-2104307798_

closed 2024-05-13 by aswaterman

davidlt, 2024-05-14:
I think this is still just partially solved. `Zbkc`, and even `Zba` (and many other) extensions continue to have:
```
[NOTE,caption=Frozen]
====
The Zba extension is frozen.
====
```
Yet it's ratified in 2021.

Also looking at instructions, e.g. `28.5.47. xnor` shows that `Zbb (Basic bit-manipulation)` `Minimum version` is `0.93`. Also `Lifecycle state` is `Frozen`.

Yet the chapter:
```
Chapter 28. "B" Extension for Bit Manipulation, Version 1.0.0
```
I don't think that `Minimum version` should list anything lower than ratified version.

wmat, 2024-05-14:
Thanks. I'll fix those.