riscv/riscv-isa-manual #1399: Provide clear SW/HW guidelines for managing the PTE cache

Wangrodman, 2024-05-13:
The PTE cache is a common hardware implementation to accelerate page table walks. However, when implementing a hypervisor extension, further specifications are desired to regulate the software and hardware's management of the PTE cache

Q1: Is it permissible for different stages to share the same PTE? For example, can a one-stage and G-stage use the same PTE?
Q2: If Q1 is permitted. When a PTE is modified, can the MMU expect software to both use sfence.vma V=0 and hfence.gvma to ensure that the latest PTE is observed?

gfavor, 2024-05-13:
On Mon, May 13, 2024 at 2:34 AM Wangrodman ***@***.***> wrote:

> The PTE cache is a common hardware implementation to accelerate page table
> walks. However, when implementing a hypervisor extension, further
> specifications are desired to regulate the software and [...]

The spec does not disallow this, so yes.  But it is software's
responsibility to manage this sharing wrt caching of PTEs.

Q2: If Q1 is permitted. When a PTE is modified, can the MMU expect software
> to both use sfence.vma V=0 and hfence.gvma to ensure that the latest PTE is
> observed?
>

Yes.  As the spec tries to convey, changes to PTEs are not guaranteed to be
seen by hardware tablewalks (and caching of those) until appropriate
sfence/hfence operations are performed.

Greg

Message ID: ***@***.***>
>

Wangrodman, 2024-05-14:
> On Mon, May 13, 2024 at 2:34 AM Wangrodman ***@***.***> wrote: The PTE cache is a common hardware implementation to accelerate page table walks. However, when implementing a hypervisor extension, [...]

Got it, no further question.

closed 2024-05-14 by Wangrodman