riscv/riscv-isa-manual #1402: Should Vector chapter version be bumped to 2.0?

wmat, 2024-05-14:
The versioning of the Vector chapter is causing some confusion.  The introduction to the chapter states that "This document is version 1.1-draft of the RISC-V vector extension.

NOTE: This version holds updates gathered after the start of the public review.  The spec will have a final update to version 2.0 at
time of ratification."

How should we proceed with versioning this chapter?

camel-cdr, 2024-05-14:
I think the original idea was to bump the spec version to 2.0 once ratified, but that idea got dropped. (Somebody correct me if I'm wrong)

Considering that RVV 1.0 is used in a lot of marketing, slides, articles, and conversations retroactively renaming it to RVV 2.0 would be very problematic.
Especially considering the history of XTheadVector ("RVV 0.7.1" based) being marketed as RVV support adding another version to the mix that doesn't change anything would just cause confusion.

I think the best course of action is to just remove that NOTE.

gfavor, 2024-05-14:
After checking with people involved with RVV development (including the TG chair - Krste), it is recognized that there are these bits of text that are a ready source of confusion - and are now being removed from the spec as part of the current ISA manual conversion/integration/cleanup efforts that are occurring.  The net of the cleanup will clearly show that the current version is v1.0.  (Also note that the text mentioning v1.1 was anticipating a new version of RVV that never happened.)

camel-cdr, 2024-06-08:
So whats the verdict?
I think this needs to be addressed, and it sounds like the solution is to remove that passage.
Is this change still in some kind of internal review?

gfavor, 2024-06-08:
The latest (integrated) vector chapter has been cleaned up to properly
reflect that it is v1.0.  Various bits of text that looked forward to
possible or probable changes that would warrant updating to v2.0 or v1.1,
have been removed since those changes never occurred.

Greg

aswaterman, 2024-06-09:
What @gfavor said. I neglected to close this issue when the spec integration occurred, but now’s as good a time as any.

closed 2024-06-09 by aswaterman