riscv/riscv-isa-manual #1403: Fix table issues

kersten1, 2024-05-14:
Right now, in the unpriv, there are 2 - table 2's. The table on page 19 and the table on page 21 are both labeled as Table 2. This can be fixed by removing the `Table 2` label from the Table id `[[trapcharacteristics, Table 2]]` to be `[[trapcharacteristics]]`.  

Also remove from Table 1 - `[[instlengthcode, Table 1]]`.

Then verify that it is fixed

Peter-Herrmann, 2024-05-27:
https://github.com/riscv/riscv-isa-manual/pull/1429 fixes these hard-coded values, as well as some linting and style fixes. The rendered docs appears correct, but are here for review: 
- [unpriv-isa-asciidoc-9f20da5.epub.zip](https://github.com/riscv/riscv-isa-manual/files/15449635/unpriv-isa-asciidoc-9f20da5.epub.zip)
- [unpriv-isa-asciidoc-9f20da5.html.zip](https://github.com/riscv/riscv-isa-manual/files/15449637/unpriv-isa-asciidoc-9f20da5.html.zip)
- [unpriv-isa-asciidoc-9f20da5.pdf.zip](https://github.com/riscv/riscv-isa-manual/files/15449638/unpriv-isa-asciidoc-9f20da5.pdf.zip)

kersten1, 2024-06-18:
This one is fixed.

closed 2024-06-18 by kersten1