riscv/riscv-isa-manual #1409: Zawrs: "These instructions are not supported in a constrained LR/SC loop."

tariqkurd-repo, 2024-05-15:
The description for Zawrs says `These instructions are not supported in a constrained LR/SC loop.`

What does this mean exactly?

Should the core detect that they are within an LR/SC loop and take an illegal exception, or ignore them, or what?

Please clarify

ved-rivos, 2024-05-15:
It means that if a `WRS.*` instruction is in a LR/SC loop then it is not a _constrained_ LR/SC loop and guarantees around eventual success of store-conditional does not apply. This instruction is not part of the set of instructions listed in section 14.3 as supported in a _constrained_ LR/SC loop.

aswaterman, 2024-05-15:
This is all just a matter of clarification, but can’t we just delete this clause altogether? The A chapter already excludes these new instructions from constrained loops by not mentioning them.

aswaterman, 2024-05-15:
No, the statement was always superfluous. The constrained LR/SC loop definition already precludes new instructions by virtue of admitting a restricted set and excluding all others.

ved-rivos, 2024-05-16:
Update in PR #1416

closed 2024-05-20 by tariqkurd-repo