riscv/riscv-isa-manual #1418: Sstc description hasn't been merged well

tariqkurd-repo, 2024-05-20:
In this chapter:

>Chapter 16. "Sstc" Extension for Supervisor-mode Timer Interrupts, Version 1.0.0

I don't think this text should be there anymore

>To make it easy to understand the deltas from the current Priv 1.11/1.12 specs, this is written as the
>actual exact changes to be made to existing paragraphs of Priv spec text (or additional paragraphs
>within the existing text).

and in these chapters:

>3.1.18. Machine Environment Configuration (menvcfg) Register
>18.2.5. Hypervisor Environment Configuration Register (henvcfg)

there is this text:

>The definition of the STCE field will be furnished by the forthcoming Sstc extension. Its allocation
within [mh]envcfg may change prior to the....

Also the `stce` bit of Xenvcfg (which I was actually looking for when I found this) _isn't_ mentioned in the `Sstc` chapter at all, so the definition of that field seems to be missing.

aswaterman, 2024-05-20:
I believe @gfavor is aware that one of the many, many things on his to-do list is to better integrate the Sstc chapter.  (Sstc is certainly not the only offender in this regard.)

ved-rivos, 2025-12-16:
Addressed by #2504

closed 2025-12-16 by aswaterman