riscv/riscv-isa-manual #1420: Does mcountinhibit Inhibit Only Unprivileged Counters, or Both Machine and Unprivileged Counters?

cebarobot, 2024-05-22:
I'm cofusing about the `mcountinhibit` CSR. Does `mcountinhibit` inhibit only Unprivileged Counters, or both Machine and Unprivileged Counters?

The Privileged ISA manual says:
> When the CY, IR, or HPMn bit in the mcountinhibit register is clear, the `cycle`, `instret`, or `hpmcountern` register increments as usual. When the CY, IR, or HPM_n_ bit is set, the corresponding counter does not increment.

It only refers to **Unprivileged Counters** (`cycle`, `instret`, or `hpmcountern`) and not mensions the **Machine Counters** (`mcycle`, `minstret`, or `mhpmcountern`) at all.

As `cycle`, `instret` and `hpmcountern` is a read-only shadow of `mcycle`, `minstret` and `mhpmcountern`, there is an intuitive understanding based on the text: "mcountinhibit` only inhibits Unprivileged Counters":
> When the CY, IR, or HPM_n_ is set, the hart makes a data copy of `mcycle`, `minstret` and `mhpmcountern` at that moment.
> Reading Unprivileged Counters returns that copy, while machine counters keep going.

In that understanding, the purpose of `mcountinhibit` (reduce energy consumption) make no sense, which leads to more confusion.

I also checked some open-source implementation, and they did as below:
* [Spike (riscv/riscv-isa-sim)](https://github.com/riscv/riscv-isa-sim): `mcountinhibit` is [read-only zero](https://github.com/riscv-software-src/riscv-isa-sim/blob/c81d8e73daf1875d79a54f28f67df150fba0e44c/riscv/processor.cc#L292).
* [BOOM](https://github.com/riscv-boom/riscv-boom): `mcountinhibit` is not implemented.
* [Rocket](https://github.com/chipsalliance/rocket-chip): `mountinhibit` will inhibit [Both Machine and Unprivileged Counters](https://github.com/chipsalliance/rocket-chip/blob/dbcb06afe1c76d1129cb6d264949322a34c37185/src/main/scala/rocket/CSR.scala#L584).
* [CVA6](https://github.com/openhwgroup/cva6): `mcountinhibit` will inhibit [Both Machine and Unprivileged Counters](https://github.com/openhwgroup/cva6/blob/26e6a8de4e5e1bf95b0044d2825aaf005a3a62cc/core/csr_regfile.sv#L814)

It seems that "inhibit both machine and unprivileged counter" is the correct answer. If so, could we add some clarification on this issue in the ISA manual?

aswaterman, 2024-05-22:
I think the ISA manual is already saying the right thing.

The spec states that "The `cycle`, `instret`, and `hpmcountern` CSRs are read-only shadows of `mcycle`, `minstret`, and `mhpmcountern`, respectively."  That means that e.g. `cycle` does not contain its own state.  So if `cycle` stops counting, `mcycle` must have stopped counting, as well.

It might've been clearer if the `mcountinhibit` spec referred to e.g. `mcycle` instead of `cycle`, but the meaning is the same either way, because of the "read-only shadow" definition.

From what you described, it sounds like all of the open-source implementations you mention are correctly interpreting the spec.

cebarobot, 2024-05-22:
> It might've been clearer if the `mcountinhibit` spec referred to e.g. `mcycle` instead of `cycle`, but the meaning is the same either way, because of the "read-only shadow" definition.

I agree with that. It's better to refer to `mcycle` in spec. I'd like to open an PR to fix this.

Also, `cycle`, `instret`, or `hpmcountern` may be non-existent if Zicntr or Zihpm is not implemented. From this perspective, it's also better to use `mcycle`, `minstret` and `mhpmcountern`.

aswaterman, 2024-05-22:
Valid point. Would you like to prepare a PR that makes the appropriate changes to this section? I think it’s just a simple search-and-replace; no new text should be needed.

aswaterman, 2024-05-22:
cc @gfavor

cebarobot, 2024-05-22:
> Valid point. Would you like to prepare a PR that makes the appropriate changes to this section? I think it’s just a simple search-and-replace; no new text should be needed.

I think it's better if you could just commit the changes on the main branch. Preparing a PR is a bit troublesome after all.

aswaterman, 2024-05-22:
I was politely trying to ask you to do the work for me. I guess I’ll try once more.

cebarobot, 2024-05-22:
> I was politely trying to ask you to do the work for me. I guess I’ll try once more.

Sorry, I just misunderstand it. I'm doing that.

aswaterman, 2024-05-22:
Thank you for doing so!

closed 2024-05-22 by aswaterman