riscv/riscv-isa-manual #1421: privileged isa spec. table 3 read-only csr's

farukyld, 2024-05-22:
are those ones correct in the `Table 3` in the document `Volume 2, Privileged Specification version 20240411`?
[image]
[image]

allenjbaum, 2024-05-22:
Is there some reason you believe that the spec has a bug in that table?

farukyld, 2024-05-22:
here:

in `Table 6`
[image]

in `Table 7`
[image]

farukyld, 2024-05-22:
and also in the previous versions of privileged spec. (I have downloaded 20211203) those ranges were specified as read-only or custom read-only.

in this document: `The RISC-V Instruction Set Manual Volume II: Privileged Architecture Document Version 20211203` in `Table 2.1` 

[image]

aswaterman, 2024-05-22:
This is a transcription error (cc @wmat).  I'll fix it.

closed 2024-07-25 by tariqkurd-repo