riscv/riscv-isa-manual #1423: What will happen if a hart executes `sinval.vma` with no fence before `sfence.w.inval`

huxuan0307, 2024-05-23:
Is some mechinism needed to handle this software mistake? If the implementation do no fence operation when excuting `sinval.vma`, is it necessary for implementation to check if there is an opening `sfence.w.inval` before `sinval.vma`.

gfavor, 2024-05-23:
In short, No and No.  It is software's responsibility to use these instructions appropriately, and (for that matter) it is its choice to use them in unexpected ways.  The architectural intent is to provide these instructiosn as the tools for software to do what it wants.  There is no expectation for hardware to catch and/or defend against what it might (correctly or incorrectly) consider to be a software bug.

huxuan0307, 2024-06-18:
Ok, thanks

closed 2024-06-18 by huxuan0307