riscv/riscv-isa-manual #1427: Publish nightly draft to RISC-V's Github Pages

VitalyAnkh, 2024-05-24:
Certainly! Here's a more polished version:

The draft HTML version of the RISC-V ISA manual is available for download on the release pages but is not currently hosted on a webpage. Could it be published on RISC-V's GitHub Pages, similar to how riscv-cheri is hosted at [riscv.github.io/riscv-cheri/](https://riscv.github.io/riscv-cheri/), to facilitate easier browsing of the latest draft?

It appears this was previously discussed in this issue: https://github.com/riscv/riscv-isa-manual/issues/175, but it has not yet been implemented as a published webpage.

Timmmm, 2024-11-04:
I have put all of the [HTML RISC-V specs here](https://riscv-specs.timhutt.co.uk/). It's updated automatically every night.

I'll try and keep the URLs stable so you can link to them.

wmat, 2024-11-04:
Hi Tim, sorry, I'm replying late.

> The problem is you can't link to an HTML file on Github releases, and
> opening it is a bit awkward (download and then open, then you end up with a
> lot of ... (1).html files.
>
> We did this for CHERI: https://riscv.github.io/riscv-cheri/
>
> It's quite easy to set up (see here
> ):
>
>
>       # Upload GitHub pages artefacts.
>       - name: Make gitlab pages directory
>         run: mkdir dist && cp build/*.html dist/index.html
>         if: github.event_name == 'push'
>
>       - name: Upload artifact
>         uses: ***@***.***
>         with:
>           path: dist
>         if: github.event_name == 'workflow_dispatch' || github.event_name == 'push'
>
> ...
>
>   # Deploy HTML to Github pages.
>   deploy:
>     if: github.event_name == 'push' && github.ref == 'refs/heads/main'
>
>     needs: build
>
>     permissions:
>       pages: write
>       id-token: write
>
>     environment:
>       name: github-pages
>       url: ${{ steps.deployment.outputs.page_url }}
>
>     runs-on: ubuntu-latest
>     steps:
>       - name: Deploy to GitHub Pages
>         id: deployment
>         uses: ***@***.***
>
> The big downside is it just deploys the latest version as the Github Pages
> page, so the URL just always points to the latest version and you can't see
> old versions.
>
> @wmat <https://github.com/wmat> what's the timeline for this
> documentation library? If it's going to be a year maybe it is worth using
> Github Pages in the interim.

The timeline is fluid but I'm hoping to have something to release by end of
year with Priv and Unpriv at minimum.

Also will you be able to link directly into the specs in the documentation
> library e.g.
> https://specs.riscv.org/riscv-isa-manual/privileged/v1.015/index.html#some_section,
> and will it have all the same draft releases as this repo?

Yes, you'll be able to link directly to the spec's sections and versioning
will match the github repo as that's the source for Antora based
documentation websites.

Timmmm, 2024-12-05:
@wmat are you planning to publish a multipage version of the spec like [this](https://riscv-software-src.github.io/riscv-unified-db/manual/html/isa/20240411/chapters/zifencei.html#:~:text=The%20FENCE.I%20instruction%20is%20used%20to%20synchronize%20the%20instruction%20and%20data%20streams) or a single page version like [this](https://riscv-specs.timhutt.co.uk/spec/20240411/unpriv-isa-asciidoc.html#zifencei:~:text=The%20FENCE.I%20instruction%20is%20used%20to%20synchronize%20the%20instruction%20and%20data%20streams) (or both)?

wmat, 2024-12-05:
The HTML will be multipage with the PDF living alongside.  Here is a very
early POC http://52.91.38.114/isa/index.html and as you can see, each
chapter is a page.

wmat, 2024-12-06:
> Took at look at the POC and it matches what @dhower-qc
> <https://github.com/dhower-qc> publishes with UDB here:
> https://riscv-software-src.github.io/riscv-unified-db/manual/html/isa/20240411
>
> So, I'll make links to that as discussed in #1397

Yes, we're both using the default Antora UI bundle so they look very
similar.  We'll be rebranding the UI to match the new riscv.org website.

aswaterman, 2025-12-10:
Whenever a PR is merged, a tag is created that contains the HTML and PDF representations of both specs.

closed 2025-12-10 by aswaterman