riscv/riscv-isa-manual #1430: Wavedrom encodings are inconsistently formatted

christian-herber-nxp, 2024-05-28:
I am failing to understand the logic behind the formatting of instruction encodings in wavedrom.
For one, it is quite inconsistent between extensions (e.g. compare C to Zc).
But even within extensions, or even a single instruction, I do not understand how the type is assigned (that translates to the background color). See e.g.:
[image]

Two parts of one immediate have different colors. One operand red, one operand green. 

Having clarity on how to format the instructions will greatly help authors of extensions to do it right from the start.

christian-herber-nxp, 2024-05-28:
@wmat : thanks, i updated the colors in https://github.com/riscv/riscv-zilsd/pull/18. 
From that, I can for sure say that the types in C need to be revisited.

gfavor, 2024-05-28:
Given that the "old" ISA manuals did not use color coding, should we drop all color coding in the new manuals?  Or should we have Bill identify one standard color coding that we will gradually convert to?

ved-rivos, 2024-05-28:
IMO: The colors are a bit distracting and don't add much. The look and feel of the "old" manual was better.

wmat, 2024-05-28:
The addition of colored backgrounds to the wavedrom diagrams predated my
joining the project.  I always assumed it was done for a reason but perhaps
not.  Removing the colorization (type attributes) from the wavedrom would
certainly make writing wavedrom that much easier.

gfavor, 2024-05-28:
Let's see if @aswaterman concurs with dropping all color coding.

christian-herber-nxp, 2024-05-28:
personally, I like the colors, if used meaningfully and consistently.
Especially in less regular encodings like C/Zc, helps to get an overview.

aswaterman, 2024-05-28:
I think the two valid resolutions are to drop the colors or to use the colors consistently.  Given the latter option is much more work, and isn't exactly of high priority, I'd be fine with dropping them.  We can always reverse course in the future if we have the bandwidth.

wmat, 2024-05-28:
I assume the colors are simply decoration, correct?

Bill Traynor
Documentation Build and Release Engineer
RISC-V International

Join us in Munich, Germany at RISC-V Summit Europe
<https://riscv-europe.org/summit/2024/> from 24-28 June. Be a part of the
new wave of European computing innovation!

aswaterman, 2024-05-29:
Right, they’re just there to improve the visual appearance and help readers see the boundaries between fields. There’s no semantic content.

kersten1, 2025-01-09:
This is done.

closed 2025-01-09 by kersten1