riscv/riscv-isa-manual #1444: Clarify the impact of two-stage translation apply pointer masking

Wangrodman, 2024-06-03:
Considering a hardware design in Sv57 with PMLEN=7, according to section 3.5, "Interaction with Two-Stage Address Translation," it appears that the MMU has an additional effort to handle masking of the MSB 2 bits of the GPA when vsatp.mode=BARE. However, regardless of whether pointer masking is enabled or not, the MSB 2 bits are always zero, and there seems to be no need for extra handling. Is my understanding correct?

gfavor, 2024-06-04:
In two-stage translation, G-stage uses Sv57x4 translation mode - which supports 59-bit GPAs.

Wangrodman, 2024-06-04:
> In two-stage translation, G-stage uses Sv57x4 translation mode - which supports 59-bit GPAs.

I understand that GPA[58:57] is within the masking range. However, I believe these two bits should always be 0, regardless of whether pointer masking is applied. If I am incorrect, please provide me with an example. I would be grateful for your help.
[image]

gfavor, 2024-06-05:
The H extension chapter very clearly specifies that Sv57x4 takes in and translates 59-bit GPAs.  Where is there anything in the arch spec that indicates that the high two msb's must be zeroes?

aswaterman, 2025-12-09:
It's as Greg wrote.