riscv/riscv-isa-manual #1446: `mhartid` wording re. hart ID zero confusing/incorrect?

TommyMurphyTM1234, 2024-06-06:
The latest release of the privilged spec:

* https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-isa-release-3f725b5-2024-05-31

has this (_emhpasis_ is mine):

> **3.1.5. Hart ID (mhartid) Register**
> 
> The mhartid CSR is an MXLEN-bit read-only register containing the integer ID of the hardware
> thread running the code. This register must be readable in any implementation. Hart IDs might not
> necessarily be numbered contiguously in a multiprocessor system, but _at least one hart must have a
> hart ID of zero_. Hart IDs must be unique within the execution environment.

Is _"at least one hart must have a hart ID of zero"_ actually correct? Can more than one hart have an ID of zero? Or is it the case that exactly one hart must have an ID of zero?

pdonahue-ventana, 2024-06-06:
The next sentence says "Hart IDs must be unique within the execution environment" which means that more than one hart cannot have an ID of zero.

TommyMurphyTM1234, 2024-06-06:
Thanks @pdonahue-ventana.

> The next sentence says "Hart IDs must be unique within the execution environment" which means that more than one hart cannot have an ID of zero.

Yes - I know - that's why I think that _"**at least** one hart must have a hart ID of zero"_ is unnecessarily confusing and should actually be  _"**exactly** one hart must have a hart ID of zero"_.  Why have one sentence specify a looser constraint only for the next sentence to further tighten it up when it can be stated more clearly, concisely and correctly in the first place/sentence?

Also - I wasn't sure if _"in a multiprocessor subsystem"_ and _"within the execution environment"_ were different contexts that mayhe had different constraints in this respect.

The wording seems unnecessarily confusing in my opinion.

pdonahue-ventana, 2024-06-06:
I had mentally swapped the original text and your proposed text.  I assume that the key lies in the fact that a multiprocessor system may have more than one execution environment, but I'll let others (who know more) address that.

steven-bellock, 2024-09-02:
Also, is this conditional on multiprocessor systems? Or, if there is a single processor / hart, must its `hartid` be exactly `0`?

TommyMurphyTM1234, 2024-09-02:
> if there is a single processor / hart, must its `hartid` be exactly `0`?

I think that this is the case given the above wording. But, as I've said, I think that the wording could be more clearly prescriptive in order to eliminate the need to infer any of the details.