riscv/riscv-isa-manual #1466: Clarification on syntactic dependencies for vector instructions

jrahmeh, 2024-06-20:
Section 18.1.2. states:

> In general, a register r other than x0 is a source register for an instruction i if any of the following hold:
>   In the opcode of i, rs1, rs2, or rs3 is set to r

For a vector instruction, if LMUL is to m with m larger than 1, then if rs1, rs2, or rs3 is set to r, then the effective source vector registers will be r, r+1, ... r + m - 1.  If we only consider r, we would not be capturing all the dependencies of a vector instruction.

aswaterman, 2024-06-20:
As the V extension spec tersely notes: `NOTE: More formal definitions required.`

But yes, dependencies on vector register operands need to include the entire vector register group, as specified by LMUL (or, more precisely, ceil(LMUL), since when LMUL < 1, the entire vector register is treated as part of the tail).

nick-knight, 2024-06-20:
Must dependence tracking be based on LMUL, or could finer-grained tracking based on `vl` be used instead?

aswaterman, 2024-06-20:
Yeah, I was only talking about what the definition of a syntactic dependency on a vector register would be.  I don't think it's the right tool for this job.

We want to propagate dependencies at the element level, not at the vector register level.  (Of course, "element level" ends up meaning "byte level", since different EEWs may be used by the producer and consumer.)  Otherwise, it becomes impractical to implement temporal vector machines.

Informally, for arithmetic ops, we want the same dependency behavior as in this pseudocode:

```
for each i in [vstart, vl-1]:
  if (!masked || v0.mask[i])
    vd[i] = vs2[i] op vs1[i]
```

IOW, dependences only propagate through active elements, and the dependence on vl and the mask is a control dependence, not a data dependence.

pdonahue-ventana, 2024-06-25:
I think that extensions like [Zilsd](https://github.com/riscv/riscv-zilsd/blob/main/zilsd.adoc) have a similar problem.

aswaterman, 2024-06-25:
Yes, though the right answer is fortunately clear-cut in that case.