riscv/riscv-isa-manual #1480: Should HLVX effect vsstatus.SUM?

Wangrodman, 2024-06-27:
The hypervisor uses HLVX to emulate guest execute code, and SUM is used to modify load and store. Does HLVX get affected by vsstatus.SUM?

gfavor, 2024-06-28:
The H spec says that HLVX differs from HLV only in that "execute permission
takes the place of read permission during address translation".  And the
spec for all the H*V* instructions says that "the explicit memory access is
done *as though in* VU-mode when SPVP=0, and *as though in* VS-mode when
SPVP=1. As usual when V=1, two-stage address translation is applied, and
the HS-level sstatus.SUM is ignored."

So the question is equivalent to whether a normal guest load is affected by
vsstatus.SUM.  Which is equivalent to whether a normal host load is
affected by sstatus.SUM.  Namely yes (in the ways defined by SUM).

Greg

Wangrodman, 2024-06-28:
Thanks for your clarification

closed 2024-06-28 by Wangrodman