riscv/riscv-isa-manual #1487: Proposal to remove RVC HINTs with rd != x0

cyyself, 2024-07-02:
The encoding of HINTs in RVC contains some instructions with rd != x0. When some CPU implements SXL or UXL change and chooses the simplest implementation to ignore this HINT, as the specification said, this implementation will change the architecture state that the sign-extends some wider GPR in narrow XLEN, causing HINTs to modify the architecture state.

To clarify this paradoxical behavior, the choices are:
1. Remove the following comments
   > This HINT encoding has been chosen so that simple implementations can ignore HINTs altogether, and instead execute a HINT as a regular computational instruction that happens not to mutate the architectural state."
2. Remove RVC HINTs with rd != x0
3. Explicitly allows an implementation to change architectural state when executing HINT instruction

I suggest option 2 because these HINTs haven't been used now, and it breaks the behavior for simple implementation, which may implement modifiable XLEN.

aswaterman, 2024-07-02:
Interesting point.

Something to keep in mind is that, for UXL < SXL, `addi x1, x1, 0` does not change architectural state from U-mode's perspective.  Similarly, for SXL < MXL, `addi x1, x1, 0` does not change architectural state from S-mode's perspective.  The architectural-state change is detectable only from higher modes with greater XLEN.

So, something along the lines of option (3) is entirely viable: we could say that, for HINT encodings, it is implementation-defined whether a non-canonical register is canonicalized.  An advantage of this approach is that it preserves a bunch of HINT space that would otherwise be destroyed to cater to the uncommon case of modifiable XLEN.  Another advantage is that the change is entirely isolated to the privileged-architecture spec.  (Recall the unprivileged-architecture spec doesn't comprehend the notion of modifiable XLEN.)

I can see the argument that option (2) is cleaner, but there are two downsides.  First, there are relatively few HINT encodings available in the 16-bit space; this change would eliminate 1/6 of the available space.  Second, some of those HINTs are designated for custom use; it is a backwards-incompatible change to eliminate custom encodings.

I'll bring this topic up with the ARC.  It could go either way.

cyyself, 2024-07-02:
> So, something along the lines of option (3) is entirely viable: we could say that, for HINT encodings, it is implementation-defined whether a non-canonical register is canonicalized. An advantage of this approach is that it preserves a bunch of HINT space that would otherwise be destroyed, only to cater to the uncommon case of modifiable XLEN. Another advantage is that the change is entirely isolated to the privileged architecture. (Recall the unprivileged architecture doesn't comprehend the notion of modifiable XLEN.)
> 

I agree. I think option (3) is better now. I am looking forward to this ARC topic.

> I can see the argument that option (2) is cleaner, but there are two downsides. First, there are relatively few HINT encodings available in the 16-bit space; this change would eliminate roughly 1/6 of the total space. Second, some of those HINTs are designated for custom use; it is a backwards-incompatible change to eliminate custom encodings.
> 
> I'll bring this topic up with the ARC. It could go either way, but I am going to recommend (3) to minimize disruption.

aswaterman, 2024-07-03:
The ARC agrees that option (3) is best.  I'll prepare a PR shortly.

PR #1494 (2024-07-03): Relax behavior of some HINTs when MXLEN > XLEN
Resolves #1487
closed 2024-07-03 by aswaterman