riscv/riscv-isa-manual #1496: For SV39 VA translation, which type of fault will be raised when setting ppn[2] to exceed maximum supported PA length?

Vivvvvvvv, 2024-07-03:
Hi, I am a bit confused about whether a page fault or an access fault should be raised when PPN[2] has more bits setted than needed. 

For SV39, bit [53:10] is used for PPN0/1/2, and bit [60:54] is for reserved. However, the maximum PA length that we supported is 36 bits, which leaves bit [53:34] "useless". Because if any bits inside [53:34] are set, the resulting PA, pte.ppn[2]*PAGESIZE, will definitely reaching the access-fault region. So one solution to us is that we treat any 1's inside the [60:34] as an page fault since we think that the pte is not legal to our design. The other solution is that we consider giving an access fault for 1's inside [53:34] and a page fault for 1's inside [60:54]. What are the original intention of the RISCV privilage architecture for this matter? Which solution is more suitable?

aswaterman, 2024-07-03:
You describe your system as having a 2^36-byte PA space, but I recommend thinking about it more abstractly: you actually have an infinitely sized PA space, of which only bytes [0, 2^36-1] happen to be valid.

Under this framework, the logical thing to do is to treat any PA >= 2^36 as an access exception.

Vivvvvvvv, 2024-07-03:
So for implementation, we should treat any 1's inside [53:34] as an access fault, and [60:54] as a page fault following the spec. Am I right on this?

aswaterman, 2024-07-03:
Yeah, that's right.

Vivvvvvvv, 2024-07-03:
Thank you so much for the clarification!

closed 2024-07-03 by Vivvvvvvv