riscv/riscv-isa-manual #1509: Clarify masking behaviors for xenvcfg bits

JamesKenneyImperas, 2024-07-08:
Some bits in `xenvcfg` registers appear to have similar but subtly different masking behaviors. Can you please confirm if this is intentional?

The specification of `menvfg.PBMTE` says this:

> When `PBMTE`=0, the implementation behaves as though `Svpbmt` were not implemented. If `Svpbmt` is not implemented, `PBMTE` is read-only zero. Furthermore, for implementations with the hypervisor extension, `henvcfg.PBMTE` is read-only zero if `menvcfg.PBMTE` is zero.

However, the specification of `menvfg.SSE` says this:

> When `SSE` field is 0, **the following rules apply to privilege modes that are less than M**:
>     The `henvcfg.SSE` and `senvcfg.SSE` fields will read as zero and are read-only.

These seem to describe different behavior when `henvcfg` and `senvcfg` are accessed **from M-mode**: the `PBMTE` bit, will be read-only zero in these two registers in that case, but the `SSE` bit will be visible as normal, and only appear hard-wired to zero when accessed in less privileged modes. Is this the intention?

Thanks.

JamesKenneyImperas, 2024-07-08:
Ok, I've reread the fundamental state enable register definition, and that says:

_For every bit in an mstateen CSR that is zero (whether read-only zero or set to zero), the same bit appears as read-only zero in the matching hstateen and sstateen CSRs._

So I'm guessing this is just sloppiness in the SSE field definition.

aswaterman, 2024-07-08:
I can't imagine the intention was for the behavior of accessing `senvcfg` or `henvcfg` to depend on privilege mode, since that would be a virtualization hole.

Given that there is no support for shadow stacks in M-mode, most of the statements in that paragraph are already degenerate.  So I'm pretty sure the intent is that the words "to privilege modes that are less than M" should simply be deleted.

@ved-rivos can you comment?

ved-rivos, 2024-07-08:
Except the `SSAMOSWAP.W/D` which will not be illegal in M-mode. Pulling the "The henvcfg.SSE and senvcfg.SSE fields will read as zero and are read-only." statement out of the bulleted list and replacing it  with "When menvcfg.SSE is 0,  henvcfg.SSE and senvcfg.SSE fields are read-only zero." would match the intended behavior. Does that look okay?

aswaterman, 2024-07-08:
Good point re: SSAMOSWAP.  That resolution works for me.  Can you PR it?

One other detail, and sorry if we already discussed this, but shouldn't SSAMOSWAP become reserved, rather than explicitly illegal?  (There might be non-conforming extensions living in the opcode when SSE=0.)

ved-rivos, 2024-07-08:
> That resolution works for me. Can you PR it?

Will do.

> One other detail, and sorry if we already discussed this, but shouldn't SSAMOSWAP become reserved, rather than explicitly illegal? (There might be non-conforming extensions living in the opcode when SSE=0.)

When Zicfiss is not implemented it is reserved and not illegal. But when Zicfiss is implemented the behavior is to cause an illegal or virtual instruction based on xenvcfg. And the corresponding instructions revert to Zimop/Zcmop behavior and CSR SSP access causing illegal/virtual instruction exception.

aswaterman, 2024-07-08:
Ack.

ved-rivos, 2024-07-08:
Submitted PR https://github.com/riscv/riscv-isa-manual/pull/1511

aswaterman, 2024-07-09:
Resolved by #1511

closed 2024-07-09 by aswaterman

JamesKenneyImperas, 2024-07-10:
Thanks for that correction.

As an aside, I wonder whether it would be worth restructuring the documentation of the `xenvcfg` register hierarchy to avoid unnecessary duplication and ambiguity? Currently, individual bits are documented in the extensions to which they apply; I think it would be better to have standard behavior documented in just one place and reference that from extensions.

By "standard" behavior, I mean something like:

1. if a bit is 0 in `menvcfg`, then it appears as 0 in `henvcfg` and `senvcfg` and the behavior implied by that bit appears to be entirely unimplemented at those privilege levels.
2. If a bit is 1 in `menvcfg` then the behavior implied by that bit is enabled for (H)S level. If the bit is writable in `senvcfg`, then `senvcfg` controls whether the behavior implied by that bit is enabled at (H)U level.
3. If a bit is 1 in `menvcfg` and 0 in `henvcfg` then the behavior implied by that bit appears to be entirely unimplemented at VS and VU levels, and the bit will appear forced to zero in `senvcfg` when viewed from VS level, irrespective of its actual value.
4.  If a bit is 1 in `menvcfg` and `henvcfg` then the behavior implied by that bit is enabled for VS level. If the bit is writable in `senvcfg`, then `senvcfg` controls whether the behavior implied by that bit is enabled at VU level.
5. If the rules above imply that an instruction controlled by the bit is illegal, the trap taken will be a _Virtual Instruction_ trap if executing in VS or VU mode when the bit is 1 in `menvcfg`. Otherwise, the trap taken will be an _Illegal Instruction_ trap.

It's quite likely that I've got the details wrong here somewhere, or that there is a much better way to word it, but I think anything we could do to avoid having to describe everything verbatim in every extension controlled by the `xenvcfg` registers would be a huge help.

aswaterman, 2024-07-10:
Note that not all envcfg bits follow this pattern (e.g. FIOM).  That isn't fundamentally problematic, but if anyone wants to try out this refactoring to see how it looks, they'd need to take the corner cases into account.