riscv/riscv-isa-manual #1512: VS-mode perform csr write operation to read-only csrs which are disabled by hstateen0 raise illegal or virtual instruction exception?

Jahhow, 2024-07-09:
(1) Normally when we perform csr write operation to read-only csrs, a illegal-instruction-exception should be raised.

(2) But with Smstateen extension, when mstateen0[59]=1, hstateen0[59]=0, AIA spec states that access stopi should raise virtual-instruction-exception:

> [image]

But since stopi is a read-only csr, so when VS-mode perform csr write operation to stopi or vstopi,  the above rule (1) also applies. In this case, which exception should be taken?

Thank you.

gfavor, 2024-07-10:
One can view rule (1) as the default rule that applies both inside an doutside of a guest VM, and then rule (2) selectively overrides the default behavior when accessing from inside a VM and for certain *stateen* bit settings.  Also note that rule (2) is not specific to access type, but applies to all accesses to a CSR.

So, in short, rule (2) applies when it is in effect.

Jahhow, 2024-07-11:
Thanks a lot 👍

closed 2024-07-11 by Jahhow

Jahhow, 2024-07-11:
Does rule (2) applies to instruction `csrw vstopi` while executing in VS/VU-mode?

gfavor, 2024-07-11:
I can't say definitively, but I believe rule (2) - since it applies to all accesses in VS/VU-mode - applies to all forms of CSR instructions that try to access a CSR (such as vstopi).

YenHaoChen, 2024-08-02:
According to the RISC-V specification, the illegal instruction exception takes precedence over the virtual instruction exception (Table 33). It is also standard practice to conduct broad CSR address validations, which include verifying privileges and read/write or read-only statuses, prior to executing specific CSR evaluations, such as *stateen checks. Consequently, it would be appropriate to trigger an illegal instruction exception in this scenario. Would you agree with this approach, @gfavor @Jahhow?
[image]

aswaterman, 2024-08-04:
The figure you quoted actually indicates that illegal and virtual instruction exceptions have the same priority level, since they are in the same cell in the table.

Informally, the rule is: if the same instruction when executed in HS-mode would have raised an illegal-instruction exception, it should raise an illegal-instruction in VS-mode.  Otherwise, it should raise a virtual-instruction exception.

Following this rule, it sounds to me like writing `stopi` should cause an illegal-instruction exception regardless of mode.