riscv/riscv-isa-manual #1520: mhpmeventNh and Sscofpmf

dhower-qc, 2024-07-11:
The current manual states:

The `mhpmevent__n__h` CSRs are provided only if the Sscofpmf extension is implemented.

Why is that the case? That would prevent an RV32 implementation from using an event > 32 bits, which isn't stated as a restriction anywhere else in the manual.

gfavor, 2024-07-11:
Before Sscofpmf came along, the achoitecture did not provide mhpmevent*h CSRs for RV32.  I can't comment about that choice made many years ago (although I suspect it was felt that 32 bits of event register space was likely sufficient for RV32 implementations).

When Sscofpmf came along, one of the goals was to minimize the risk of (and/or actual) infringing on existing implementations' use of the existing event register bits - with the idea that the high bits were the safest to use.  But given the more limited number of existing event bist in RV32, it was decided that it was better to introduce *h registers instead of "stealing" bits from the upper parts of the existing 32 event register bits (which also allows for other future extensions that want to introduce more bits up high and out of the way of uses of all the lower and middle bits).

dhower-qc, 2024-07-11:
got it, thanks

closed 2024-07-11 by dhower-qc

aswaterman, 2024-07-11:
> I can't comment about that choice made many years ago (although I suspect it was felt that 32 bits of event register space was likely sufficient for RV32 implementations).

For the historical record, Greg's suspicion is right; we simply didn't foresee the need for >32b hpmevent registers, and so we didn't go out of our way define another group of CSRs.