riscv/riscv-isa-manual #1528: how to understand bit 63 in mstateen0 is defined to control access to the matching sstateen and hstateen CSRs?

ZeyueShen, 2024-07-13:
In spec：
https://github.com/riscvarchive/riscv-state-enable/blob/main/content.adoc
[image]

It describes that access to hstateen0 csr is controlled by mstateen0[63], but does not specify whether failure to access will result in an exception. Similarly, access to sstateen0 csr is controlled by hstateen0[63], and there is no mention of non-access handling.

But in spike:
[image]

It means that when mstateen0[63]  is zero, access to hstateen0 in privileged mode less than m mode will cause an illegal instruction exception.
[image]

It means that when mstateen0[63] &  hstateen0[63] is zero, accessing sstateen0 in virtualization mode causes a virtual instruction exception.

So can you describe in detail the controlled access of hstateen0 and sstateen0? I am looking forward to your answer!

gfavor, 2024-07-16:
The spec (section 4.1) says:

"when a stateen CSR prevents access to state by less privileged levels, an attempt in one of those privilege modes to execute an instruction that would read or write the protected state raises an illegal instruction exception, or, if executing in VS or VU mode
and the circumstances for a virtual instruction exception apply, raises a virtual instruction exception
instead of an illegal instruction exception."

ZeyueShen, 2024-07-17:
ok, Thanks for your reply!

senlinhe, 2025-06-23:
But this appears inconsistent with the explanation here.

senlinhe, 2025-06-24:
[image]

gfavor, 2025-06-24:
That is the section 5.5. text, but what is the inconsistent text you alluded to from an "earlier chapter"?

senlinhe, 2025-06-25:
the chapter 4.1 says that if executing in VS or VU mode, a virtual-instruction should apply when stateen CSRs set to 0

[image]

for the explanation in section 5.5 text, it is "when mstateen0[60] = 0, an attempt to access one of these CSRs from a privileged mode less privileged than M-mode results in an illegal-instrucion exception." 
According to chapter 4.1, I understand when a stateen CSR prevents access to a state by less-privilege mode, We need to determine whether to report an 'illegal instruction' or 'virtual instruction' exception based on the privilege mode, but according to the explanation of Chapter 5.5 text, it seems when mstateen0[60]=0, all privilege mode less than M-mode attempt to access those protected state should result in illegal instruction. that is why I get confused