riscv/riscv-isa-manual #1529: Mnemonic for FCVT

tianboh, 2024-07-15:
I noticed that FCVT can convert from float to integer or vice versa. However, I am confused about its suffix. For example, SPEC 20.7 points out

>FCVT.W.S or FCVT.L.S converts a floating-point number in floating-point register rs1 to a signed 32-bit or 64-bit integer, respectively, in integer register rd. For XLEN > 32, FCVT.W[U].S sign-extends the 32-bit result to the destination register width.

When XLEN = 64, it seems that `FCVT.W.S` and `FCVT.L.S` both convert a 32-bit single-precision value to a sign-extends 64-bit integer. What's the difference between these two?

topperc, 2024-07-15:
> I noticed that FCVT can convert from float to integer or vice versa. However, I am confused about its suffix. For example, SPEC 20.7 points out
>
>
>
>
>
>
> When XLEN = 64, it seems that `FCVT.W.S` and [...]

When XLEN=64, FCVT.W.S converts a floating-point number to a signed 32-bit integer. FCVT.L.S converts a floating-point number to a 64-bit integer without sign extension.

tianboh, 2024-07-15:
> When XLEN=64, FCVT.W.S converts a floating-point number to a signed 32-bit integer. FCVT.L.S converts a floating-point number to a 64-bit integer without sign extension.

Thank you for your reply! Nice to see your answer again :-) In this case, what's the difference between `FCVT.W.S` and `FCVT.LU.S`? 

I think I am kind of confused for the suffix `S`. From my understanding it means `single-precision`. But I noticed the following table.

[image]

It seems that `FCVT.L.S` valid input can be a `double`. If so, it can help to answer the previous quetsion: `FCVT.L.S` converts a double to a 64-bit integer while `FCVT.W.S` converts a single to 32-bit integer(or signed 64-bit integer when XLEN = 64).

tianboh, 2024-07-15:
Oh wait, I guess a single can cover the range from $-2^{63}$ to $2^{63} -1$. So `S` should be `single-precision`.

closed 2024-07-15 by tianboh