riscv/riscv-isa-manual #1530: What is the source of NMI specified by Smrnmi extension?

zhangdujiao, 2024-07-15:
Can any interrupt (WSI, MSI, IPI) supported by the AIA architecture in the SoC be regarded as an NMI? 
Or, whether the interrupt is independent of the AIA architecture (like timer interrupt) and directly accesses into the hart by GPIO?

gfavor, 2024-07-15:
No to the first question.    See section 3.5 of the Priv manual for a baseline description of non-maskable interrupts.

zhangdujiao, 2024-07-17:
> No to the first question. See section 3.5 of the Priv manual for a baseline description of non-maskable interrupts.

Thx, I didn't find the definitions of how NMI is implemented in SoC from the specification. 
Is it completely depends on implementation? Is there any suggested implementation method? such as wire to the cores?

aswaterman, 2024-07-17:
The only thing the Smrnmi spec says on the matter is "The rnmi interrupt signals are inputs to the hart."  What drives those signals is implementation-dependent and might be defined by a platform spec.

closed 2024-07-29 by zhangdujiao