riscv/riscv-isa-manual #1533: Clarify Zicfiss interaction with Zjpm

JamesKenneyImperas, 2024-07-15:
Zicfiss adds new instructions for loads and stores to the shadow stack. Are these instructions subject to pointer masking as defined by Zjpm? If so, can these instructions be added to the list of affected instructions in the Zjpm specification, or explicitly indicated as _not_ affected by pointer masking?

Thanks.

ved-rivos, 2024-07-15:
Instructions provided by Zicfiss make explicit memory accesses and pointer masking applies to those explicit memory accesses. The `ssp` CSR is specified as being XLEN wide and no pointer masking operations are applied when software reads/writes the `ssp` CSR - either using a Zicsr instructions or using `ssrdp` instruction. 

This follows the two general rules specified in section 2.6 of the pointer masking specification - a) pointer masking applies to all explicit memory accesses b) No pointer masking operations are applied when software reads/writes to CSRs.

The pointer masking spec may have been better maintainable if it only listed exceptions to these rules than listing all instructions that confirm to the rule.

If we retain this list then `sspush/c.sspush`, `sspopchk/c.sspopchk`, and `ssamoswap` will be in the list as instructions that perform explicit memory accesses.

ved-rivos, 2024-09-27:
clarified by PR #1657

closed 2024-09-27 by aswaterman