riscv/riscv-isa-manual #1534: Clarify interaction of mstatus.MPRV and Zicfiss

JamesKenneyImperas, 2024-07-15:
Can you please clarify the interaction of `mstatus.MPRV` with Zicfiss?

The Privileged Specification says:
_When `MPRV=1`, load and store memory addresses are translated and protected, and endianness is applied, as though the current privilege mode were set to `MPP`._

And the Zicfiss specification has this kind of text for the new MOP instructions:
_The `SSPUSH` and `C.SSPUSH` instructions perform a store identically to the existing store instructions, with the difference that the base is implicitly `ssp` and the width is implicitly XLEN._

This implies to me that instructions such as `SSPUSH` can be executed in M-mode if `mstatus.MPRV` indicates a lower-level mode with Zicfiss enabled. However, the Zicfiss specification also says:
_Activating Zicfiss in M-mode is currently not supported._

Is it intended that `mstatus.MPRV` enables use of Zicfiss instructions in M-mode, if they are enabled at the target privilege level implied by `mstatus.MPRV`? If so, I think it would clarify things to state this explicitly and adjust the wording of that last sentence to _Activating Zicfiss for M-mode loads and stores is currently not supported_, or something like that.

Thanks.

ved-rivos, 2024-07-15:
> Activating Zicfiss in M-mode is currently not supported.

That is correct. When `sspush` and `sspopchk` (and their compressed equivalent) are executed at M-mode, these instructions execute as MOPs (Zimop or Zcmop) and thus do not make a load or store and hence `MPRV` does not affect them. The `ssamoswap` is not a MOP. When executed in M-mode, it performs a memory access using the effective privilege mode as determined by `MPRV` and `MPP`.

JamesKenneyImperas, 2024-07-15:
Thanks for that - closing.

closed 2024-07-15 by JamesKenneyImperas