riscv/riscv-isa-manual #1545: What are the expected default value for vector csrs (vl, vtype, etc)?

Vivvvvvvv, 2024-07-18:
Hi, it seems that the spec does not specify the default value for vector csrs such as vl and vtype. I have a scenario that the value for vl is ambiguous to me. Let's say I have following asm code: (nothing has been set before this) 

li x3, 0x8
csrwi vl, x3 (this should cause an illegal instruction, since vl is RO)
vsetvli x0, x0, e16, m1, tu, mu

For vsetvli instruction, setting rd and rs1 to x0 means that vl will keep the original value. However, the value for vl has not been set. So in this case, what value should vl holds as there is no exisiting value for vl? Should it sets the vill bits directly as SEW/LMUL ratio changed as vtype has also not been set?

gfavor, 2024-07-18:
Since there are no architecturally defined reset values for these
registers, they have UNSPECIFIED values until firmware/software initializes
them.  (Or a given hardware implementation may choose to initialize them to
implementation-specific values during hardware reset.)

kasanovic, 2024-07-18:
More generally, the execution environment interface (of which application ABI is one example) is the one providing the guarantees on initial values of registers.

closed 2024-07-18 by kasanovic