riscv/riscv-isa-manual #1550: Typos found

AFOliveira, 2024-07-22:
While reading the ISA manuals I came across with two typos in distinct parts of the manual.

### Typo 1

In the unprivileged manual, Section 39.8, page 634, path: src/naming.adoc there is a typo in the note "Many augmentations to the hypervisor-level **archtecture**"

### Typo 2

In the privileged manual, Section 19.2.9, page 152, path: src/hypervisor.adoc there is this sentence," `htinst` is a *WARL* register that **need only** be able to hold the values that the implementation may automatically write to it on a trap. " which I believe would be better written as " `htinst` is a *WARL* register that **only needs to** be able to hold the values that the implementation may automatically write to it on a trap. "

wmat, 2024-07-22:
Please submit a Pull Request with your fixes.

somyadashora, 2024-07-23:
Didn't want to raise a new issue adding other typo's here:

https://github.com/riscv/riscv-isa-manual/blob/2b35ff8cfae7630e8e565bc910698d951604a03f/src/hypervisor.adoc?plain=1#L2484

Should be "ps**e**udoinstructions"

AFOliveira, 2024-07-23:
Hello!
I'm just now submitting the PR, is it ok if I add yours @somyadashora?

PR #1552 (2024-07-23): Typos fix in naming.adoc and hypervisor.adoc
Fixes #1550
closed 2024-07-23 by wmat