riscv/riscv-isa-manual #1565: [Smrnmi] Handling delegated exception when mnstatus.NMIE=0

shubham-333, 2024-07-25:
> If the hart encounters an exception while executing in M-mode with the mnstatus.NMIE bit clear, the actions taken are the same as if the exception had occurred while mnstatus.NMIE were set, except that the program counter is set to the RNMI exception trap handler address (rather than the address specified by mtvec).

Given `mnstatus.NMIE=0`:
If the hart encounters a delegated exception in S-mode, which is supposed to trap to S-mode. What should the program counter be set to?
- address specified by `stvec`
- RNMI exception trap handler address (handle in M-mode)
- RNMI exception trap handler address (handle in S-mode)

ved-rivos, 2024-07-25:
The quoted text applies to an exception while executing in M-mode.

allenjbaum, 2024-07-25:
Right - but the question is: what happens if Smode, delegated, NMIE=0.
There really should be a spec table with all 8 (or 12) variations, which
would make this obvious - Sail needs the equivalent anyway.