riscv/riscv-isa-manual #1566: Zicfiss extension and address translation caches

JamesKenneyImperas, 2024-07-25:
The Zicfiss specification does not say anything about the effect of enabling or disabling the extension on address translation cache structures as far as I can see. It describes the effect on entries in _page tables_, which I'm assuming means the memory-resident structures only. Could you please clarify this?

I'm thinking of this scenario:
1. Non-bare-mode address translation is enabled in S mode.
2. Zicfiss is enabled for S mode.
3. An address in a shadow stack page is read in S mode. This causes an entry for that shadow stack page to be entered in a translation cache.
4. Zicfiss is disabled.
5. An address in the same (cached) shadow stack page is read.

Does that read cause an Access Fault, or does it succeed, or is this implementation-defined? Is there any requirement to use `SFENCE.VMA` on shadow stack page addresses when enabling/disabling Zicfiss for correct behavior?

Thanks.

ved-rivos, 2024-07-25:
Section 20.2.2 has a note about this. Changes to xSSE take effect immediately and address-translation caches need not be synchronized with a memory-management fence. In this scenario when Zicfiss is disabled the PTE encoding XWR=010 is reserved and any access to memory mapped with this encoding leads to a page fault exception.

JamesKenneyImperas, 2024-07-25:
Ok, thanks. Maybe I'm looking at a stale specification because I don't see that in my version. Closing.

closed 2024-07-25 by JamesKenneyImperas