riscv/riscv-isa-manual #1588: Unused wavedrom files

kersten1, 2024-08-08:
The following wavedrom files are not referenced anywhere in the isa. Delete them?

fnmaddsub.edn
cr-register-new.edn
half-store.edn
hint-nopv_rv32i.edn
hint-nopv_rv64i.edn
immediate_variants2.adoc
nop-v.edn
sploat2.adoc
zifencei-fetch.edn

These 2 are referenced but are in commented out sections:
cr-register.edn (commented out)
zihintpause-hint.edn (commented out)

Some were renamed to .edn before I realized that they weren't referenced, hence the mix of adoc and edn files. I did, however, search on just the name and not the file extension :)

aswaterman, 2024-08-08:
We should delete them.  (They'll live on in git history, anyway.)

kersten1, 2024-08-12:
Deleted all but the 2 that are still referenced but commented out.