riscv/riscv-isa-manual #1589: How to clear the pending of VS-level timer interrupts which pending is set by hvip.vstip and trap to VS mode?

ZeyueShen, 2024-08-08:
Perform the following configurations in HS mode: hvip.vstip=1, hideleg[6]=1, hie.vstie = 1,  vsstatus.sie =1. Then excuting sret to VS mode,   a VS-level timer interruprt will be excuted and trap to VS mode, but access to sip.stip(really vsip.stip) is read-only at this time, how to clear pending in VS mode without using HS mode (because it is thought that if an ecall instruction  is executed in handle for trapping to HS mode, which is considered visible in virtual machine). It confused me.

jhauser-us, 2024-08-15:
The RISC-V Privileged ISA specification, in chapter "Supervisor-Level ISA", subsection "Supervisor Interrupt Registers (`sip` and `sie`)", says:
> If implemented, STIP is read-only in `sip`, and is set and cleared by the execution environment.

This statement is modified by extension Sstc, which says
> If implemented, STIP is read-only in `sip`, and is either set and cleared by the execution environment (if `stimecmp` is not implemented), or reflects the timer interrupt signal resulting from `stimecmp` (if `stimecmp` is implemented).

Hence, to clear bit STIP in `sip`, a supervisor-level execution environment must either make an environment call (ECALL) or, if extension Sstc is implemented, write to CSR `stimecmp`.  This is consistently true for all S-level environments, whether S-mode, HS-mode, or VS-mode.

ZeyueShen, 2024-09-06:
Ok, thank you for your answer. I understand.

closed 2024-09-06 by ZeyueShen