riscv/riscv-isa-manual #1597: Clarification on tail elements for vector load fault-only-first

jrahmeh, 2024-08-12:
If a vle8ff.v encountered an exception on an element besides the first element and reduced VL, then for the purpose of identifying and processing the tail elements, do we use the reduced VL or the original VL?

aswaterman, 2024-08-12:
The spec states that "fault-only-first load instructions may update active destination elements past the element that causes trimming of the vector length (but not past the original vector length)"  and that "the values of these spurious updates do not have to correspond to the values in memory at the addressed memory locations."

This policy admits a superset of the behaviors that would be legal when treating the elements between [newVL, oldVL-1] as part of the tail.  (For example, it would be legal to write those elements with all 1s regardless of the `vta` setting.)  Therefore, it's valid for an implementation to treat those elements as part of the tail, but that isn't the only possible implementation.

jrahmeh, 2024-08-13:
Thanks for the clarification.  Closing.

closed 2024-08-13 by jrahmeh