riscv/riscv-isa-manual #1605: How to get the size of physical address and virtual address in RISCV?

yorange1, 2024-08-20:
Hello everyone，I recently saw that the cpuinfo in Linux for RISC-V architecture does not include the number of physical and virtual address bits supported by the current CPU like X86.
[image]
I found that the X86 CPU can use the cpuid instruction to obtain the physical address and virtual address information supported by the CPU,
[image]
but there seems to be no similar instruction in RISC-V. Will we add similar instructions in the future or use other methods to pass this information?

aswaterman, 2024-08-20:
The OS can figure out which virtual address sizes are supported by attempting to program `satp` and seeing if the write succeeds.  It can then remember this information in a variable in memory.

Probing for the physical address size is trickier; it requires filling in a PTE and seeing if accessing it raises an exception.  It's still doable, though.  But this information isn't really useful, anyway.  The important thing is that the OS knows where memory is located.  (This might come from e.g. the devicetree.)

kersten1, 2024-09-04:
Is this something that we want to document in guides? maybe a tip?

aswaterman, 2024-09-04:
In practice, I don't think that knowing the maximum physical address is very useful; the more important thing is for the OS to know what devices live at which addresses.  That one's a platform issue more than it is an ISA issue.

yorange1, 2024-09-05:
> The OS can figure out which virtual address sizes are supported by attempting to program `satp` and seeing if the write succeeds. It can then remember this information in a variable in memory.
> 
> The size of a physical address isn't really interesting; the important thing is that the OS knows which physical addresses contain which memories, devices, etc. (This information might come from devicetree.) The physical address width is immaterial.

Thanks for your reply. Can I assume that RISCV currently has no intention to add an instruction like X86's ```cpuid``` to obtain address width information? In addition, do we have some Spec-specified method to obtain the address of the device, or is it just determined by the platform (such as device tree)?

aswaterman, 2024-09-05:
AFAIK, there are no current plans to add a CPUID-like instruction.  The memory map is a platform matter, rather than an ISA matter, and yes, the platform might say something like use devicetree.

gfavor, 2024-09-05:
Adding to Andrew's response, a few years ago the addition of a CPUID-like instruction was explored, but ultimately the plan was to develop what is called the Unified Discovery extension for RISC-V standard "low-level" discovery (e.g. by M-mode firmware).  This information can then be used, for example, to populate industry-standard Device Tree or ACPI structures that are passed to an OS.  As far as discovery between an OS/hypervisor and user applications, that would be based on standard OS mechanisms such as ones in Linux.

kersten1, 2024-09-26:
Closing this issue then. Please reopen if needed.

closed 2024-09-26 by kersten1