riscv/riscv-isa-manual #1607: Formulas (images) not displaced in Apple Books v6.4

kbroch-rivosinc, 2024-08-21:
This may be related to the note in the README: https://github.com/riscv/riscv-isa-manual/blob/main/README.md#images-not-rendered-for-epub-files

Although it looks like the `:data-uri:` is uncommented already: https://github.com/riscv/riscv-isa-manual/blob/main/src/riscv-unprivileged.adoc?plain=1#L45

Here's what Apple MacOS (v14.6.1) Books (v6.4) presents:

[image]

kbroch-rivosinc, 2024-08-21:
Workaround was to install [calibre](https://calibre-ebook.com/) and use that "E-book Viewer" provided instead:

[image]

TommyMurphyTM1234, 2024-09-03:
The title of this issue should probably be generalised given that this problem is not specific to Apple Books and many (most?) ePub readers seem to have problems with such images - e.g. 

- https://github.com/riscv/riscv-isa-manual/issues/1621

Also - perhaps this section should be updated to reflect the fact that many (most?) ePub apps/applications have the same problem?

- https://github.com/riscv/riscv-isa-manual?tab=readme-ov-file#images-not-rendered-for-epub-files

wmat, 2024-09-03:
Looks like this is a known issue on Mac: https://github.com/asciidoctor/asciidoctor-epub3/issues/439

wmat, 2024-09-03:
This may have something to do with our usage of multiple images directories in the document header, i.e. ../docs-resources/images and /src/images.  I'm looking into this.