riscv/riscv-isa-manual #1622: Clarify presence of mtval2

JamesKenneyImperas, 2024-09-03:
The Hypervisor extension has this text:

> The hypervisor extension augments or modifies machine CSRs mstatus, mstatush, mideleg, mip, and mie, **and adds CSRs mtval2 and mtinst**.

I think this is no longer strictly accurate, because it implies that `mtval2` is present only if the Hypervisor extension is implemented. I think now it can be present if `Ssdbltrap` is present but the Hypervisor extension is absent. Perhaps a little polishing required?

Thanks.

aswaterman, 2024-09-03:
I'm struggling to see how the text you quoted implies the register is only there if the H extension is implemented.  The general understanding is that multiple extensions can add the same instruction/CSR/feature; we take the union of these things.

JamesKenneyImperas, 2024-09-03:
Hmm, ok, that isn't what I'm understanding from reading this. Perhaps we are two nations divided by a common language and all that. I'll close this and move on.

closed 2024-09-03 by JamesKenneyImperas