riscv/riscv-isa-manual #1623: Clarify xstatus.SDT and xenvcfg.DTE interactions

JamesKenneyImperas, 2024-09-03:
Ssdbltrp introduces new `xstatus.SDT` and `xenvcfg.DTE` fields. It's not clear to me how these behave when viewed and updated in M-mode via M-mode CSRs. Can you please clarify some details? Thanks.

1. Is the visible value of `mstatus.SDT` affected in any way by `menvcfg.DTE`? I'm assuming it isn't affected, because I think the purpose of `menvcfg.DTE` is to control visibility at lower exception levels. Is this right?
2. Assuming `menvcfg.DTE` doesn't affect the value of `mstatus.SDT`, Does this mean the value of `mstatus.SDT` can be freely changed to 1 or 0 by explicit writes to CSR `mstatus` irrespective of the value of `menvcfg.DTE`?
3. If a CSR write sets `mstatus.SDT` to 1, does this prevent `mstatus.SIE` being set to 1? Is this affected in any way by `menvcfg.DTE`?
4. If a CSR write to `mstatus` can set both `SDT` and `SIE` to 1, does the `sstatus` register also appear to have both bits set when viewed from M-mode (without returning to a lower exception level in the interim)?

Thanks.

ved-rivos, 2024-09-03:
1. `menvcfg.DTE` does not affect the operation at M-mode.
2. The value of `mstatus.SDT` can be changed to 1 or 0 and this is not affected by `menvcfg.DTE`.
3. `mstatus.SIE` cannot be set to 1 if `mstatus.SDT` is 1. Setting `mstatus.SDT` to 1 clears `mstatus.SIE`. `mstatus.SIE` can be set to 1 only if `mstatus.SDT` is already 0 or will be set to 0 by the same write.
4. `mstatus.SIE` cannot be 1 when `mstatus.SDT` is also one. This applies to these bits as viewed using `sstatus` as well.

closed 2024-09-04 by JamesKenneyImperas

Dark-Song, 2024-11-07:
> 1. `menvcfg.DTE` does not affect the operation at M-mode.
> 2. The value of `mstatus.SDT` can be changed to 1 or 0 and this is not affected by `menvcfg.DTE`.
> 3. `mstatus.SIE` cannot be set to 1 if [...]

It's confusing that the value of `mstatus.SDT` is not affected by `menvcfg.DTE`. The spec states that when `menvcfg.DTE` is zero, `sstatus.SDT` is read-only zero. Are `mstatus.SDT` and `sstatus.SDT` not the same bit? I thought sstatus is just a restricted view of mstatus in S mode and `mstatus.SDT` and `sstatus.SDT` are physically the same bit.

ved-rivos, 2024-11-07:
The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

SongWYE, 2024-11-10:
> The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

Thanks for your reply. Does your reply mean that, when menvcfg.DTE is 0, mstatus.SDT and sstatus.SDT still behave the same, but the read value of sstatus.SDT will be masked by menvcfg.DTE? 

I just don't understand the point why mstatus.SDT is designed to still be writable when sstatus.SDT is read-only 0. It's sstatus.SDT that controls S mode's double trap logic, so When sstatus.SDT is read-only 0, Ssdbltrp is effectively disabled, and there seems to be no meaning in having a writable SDT in M mode.

SongWYE, 2024-11-10:
> The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

Thanks for your reply. Does your reply mean that, when menvcfg.DTE is 0, mstatus.SDT and sstatus.SDT still behave the same, but the read value of sstatus.SDT will be masked by menvcfg.DTE?

I just don't understand the point why mstatus.SDT is designed to still be writable when sstatus.SDT is read-only 0. It's sstatus.SDT that controls S mode's double trap logic, so When sstatus.SDT is read-only 0, Ssdbltrp is effectively disabled, and there seems to be no meaning in having a writable SDT in mstatus.

SongWYE, 2024-11-10:
> The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

Thanks for your reply. Does your reply mean that, when `menvcfg.DTE` is 0, `mstatus.SDT` and `sstatus.SDT` still behave the same, but the read value of `sstatus.SDT` will be masked by `menvcfg.DTE`?

I'm just confused why `mstatus.SDT` is designed to still be writable when `sstatus.SDT` is read-only 0. It's `sstatus.SDT` that controls S mode's double trap logic, so When `sstatus.SDT` is read-only 0, Ssdbltrp is effectively disabled, and there seems to be no meaning in having a writable SDT in mstatus.

SongWYE, 2024-11-11:
> The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

I also want to know if `mstatus.SDT` will be updated by hardware on entering a trap if `menvcfg.DTE` is 0. The spec only states that `sstatus.SDT` will be updated by hardware on entering a trap. However, when `menvcfg.DTE` is 0, `sstatus.SDT` is read-only 0 but `mstatus.SDT` is writable. Could `mstatus.SDT` still be updated by hardware in such situation?

SongWYE, 2024-11-12:
> The answer to question to should be read in the context of the question. The `mstatus.SDT` and `sstatus.SDT` are the same bits, but the question was about the behavior of accessing using `mstatus` vs. `sstatus`.

As what I understand, `menvcfg.DTE` serves as the enable bit for Ssdbltrp extension. So when `menvcfg.DTE` is 0, the processor should behave the same as the Ssdbltrp extension is not implemented. The mstatus.SDT is introduced by the the Ssdbltrp extension, so why can it still be writable ?

ved-rivos, 2024-11-12:
> I have used open-source ISS spike to test this situation, and find out that when menvcfg.DTE is 0, sstatus.SDT is read-only 0 and mstatus.SDT can be written to 1, and sstatus.SIE cannot be written to 1 due to mstatus.SDT being 1. 

That complies with the specification.

The `menvcfg` register controls behavior of the execution environment for modes less privileged than M. When `menvcfg.DTE` is set to 0, the implementation behaves as if Ssdbltrp is not implemented for modes less privileged than M. It does not affect the privilege mode M itself.

SongWYE, 2024-11-13:
> That complies with the specification.
>
> The `menvcfg` register controls behavior of the execution environment for modes less privileged than M. When `menvcfg.DTE` is set to 0, the implementation [...]

Thank you very much for your answer. I now realize what I have misunderstood before. But I still have one last confusion about `mstatus.SIE`:
if `menvcfg.DTE` is 0, `mstatus.SDT` is written to 1 and `sstatus.SDT` is read-only 0, a write to sstatus in S mode should be able to successfully set sstatus.SIE to 1. But as Ssdbltrp is still implemented for M mode, `mstatus.SIE` cannot be set to 1 because `mstatus.SDT` is already written to 1. Does this mean that `mstatus.SIE` and `sstatus.SIE` can have different values? If so, will `mstatus.SIE` control S mode global interrupt enable?

ved-rivos, 2024-11-14:
When `menvcfg.DTE` is 0, and privilege mode is less then M, `SDT` is read only zero and implementation behaves as if Ssdbltrp is not implemented and `sstatus.SIE`, which is same as `mstatus.SIE` can be set to 1.

SongWYE, 2024-11-15:
> When `menvcfg.DTE` is 0, and privilege mode is less then M, `SDT` is read only zero and implementation behaves as if Ssdbltrp is not implemented and `sstatus.SIE`, which is same as `mstatus.SIE` can be set to 1.

The very first answer of yours in this ISSUE said：
> 4. `mstatus.SIE` cannot be 1 when `mstatus.SDT` is also one. This applies to these bits as viewed using sstatus as well.

When `menvcfg.DTE` is 0, `mstatus.SDT` is writable, so it can be set to 1, and then prevents`mstatus.SIE` to be set to 1. That is why I doubt whether `mstatus.SIE` and `sstatus.SIE` are still the same, because that means Ssdbltrp still have visible effect on S mode even if `menvcfg.DTE` is 0.

DavidYu360, 2026-03-26:
https://github.com/riscv/riscv-isa-manual/blob/0630a35b287e830f5af9b791c3b0890d61732520/src/priv/machine.adoc?plain=1#L2349-L2352
According to the above words, I think SDT in sstatus or mstatus can not set to 1 when menvcfg.DTE is zero.