riscv/riscv-isa-manual #1627: Clarify Smdbltrp details

JamesKenneyImperas, 2024-09-04:
Can I please clarify some details of the Smdbltrp extension?

1. One note states that the RNMI handler is not provided with information in `mtval2`, but I see no other reference to that CSR in this extension. Is that note incorrect?
2. When the Ssdbltrp extension causes a double-trap promoted from [H]S/VS mode with code 16, is it the case that this double-trap can cause a second double-trap to the RNMI handler with reported `mncause`=16, if `mstatus.MDT`=1? In other words, can an Smdbltrp extension double-trap itself be a double-trap from Smdbltrp perspective?

Thanks.

JamesKenneyImperas, 2024-09-04:
Answering my own question 2: no, because it isn't possible to execute in S-mode with mstatus.MDT=1.

ved-rivos, 2024-09-04:
> One note states that the RNMI handler is not provided with information in mtval2, but I see no other reference to that CSR in this extension. Is that note incorrect?

Consider a unexpected trap that occurs when executing in M-mode and say the unexpected trap was a guest page fault. If Smrnmi is implemented then the guest page fault does not write the mtval or the mtval2 - instead it writes mnepc with what would be written to mepc and mncause with what would be written to mcause. The RNMI handler can see that mncause indicates its a guest page fault but it cannot use mtval2 to determine the GPA. It would need to decode the instruction at mnepc an use its operands to determine this information.

JamesKenneyImperas, 2024-09-05:
I see. Thanks.

closed 2024-09-05 by JamesKenneyImperas