riscv/riscv-isa-manual #1632: Zip and Unzip descriptions are reversed

wmat, 2024-09-06:
As reported on reddit "In the standard page 267 the description of the zip instruction doesn't match the operation, which is correct?" by PColim in the risc-v subreddit.  This applies to the 20240811 published version of unpriv.

aswaterman, 2024-09-06:
@kdockser I'm having deja vu; I thought we already resolved this.

It looks like the SAIL code is correct but the descriptions are swapped.  Can you confirm and either file a PR or let me know so I can do so?

kdockser, 2024-09-07:
As I recall, no one was sure how to modify and rebuild the scalar crypto spec, so fixing errors like this kept getting put off.
Please go ahead and file a PR for this (and any other outstanding fixes for Scalar crypto).
Thx

pierrecolim, 2024-09-07:
Not only the description but also the encoding is incorrect, wording should be the same as in the ratified paper ( https://github.com/riscv/riscv-crypto/releases/download/v1.0.1-scalar/riscv-crypto-spec-scalar-v1.0.1.pdf )

aswaterman, 2024-09-07:
@pierrecolim could you file a PR?

pierrecolim, 2024-09-07:
no sorry, i never used github, i only have an account to open issues and comment

aswaterman, 2024-09-09:
OK, done.  LMK if you spot any further issues @kdockser @pierrecolim https://github.com/riscv/riscv-isa-manual/pull/1635

PR #1635 (2024-09-09): Fix descriptions/encodings of zip/unzip instructions
Something went wrong when these specs were integrated into the main ISA manual.  This PR restores the definitions of these instructions to the ratified ones.

Resolves #1632
commit 698e64ae (2024-09-09): Fix descriptions/encodings of zip/unzip instructions (#1635)

Something went wrong when these specs were integrated into the main ISA
manual.  This PR restores the definitions of these instructions to the
ratified ones.

Resolves #1632
closed 2024-09-09 by aswaterman