riscv/riscv-isa-manual #1656: Is there an issue, a smaller value is updated to mtime register after software doing PTP like linux?

lixiaogangkkk, 2024-09-27:
Situation: As privilege spec mentioned, mtime is read-write memory-mapped register which also works at a RTC. so it's possible that software can do PTP and update mtime periodically.
Question: If smaller value is returned in PTP, how would mtime to be updated? If mtime is updated immeidately, it would affects the read of time CSR from cores, which violate the definition and requirement in https://github.com/riscv/riscv-isa-manual/issues/1113.

gfavor, 2024-09-27:
Updating mtime, whenever one does it, is going to be disruptive to the system, especially if jumping backwards in time by more than a very small amount.  And when firmware changes or jumps mtime in any material way, that is change is going to propagate to all the time CSRs and result in a delayed change or jump in time values.

Either that needs to be avoided or needs to be done "carefully" enough such that there won't be any serious impacts.  Jumping the timebase backwards that system software is using, irrespective of whether that is an RTC, or mtime, or something else that isn't RISC-V-specific, presents the same problem.  And ditto for systems based on other architectures.

Now as far as the monotonicity and cross-hart synchronization requirements regarding RISC-V 'time', these will be inherently and unavoidably violated when software changes or jumps mtime (and indirectly all the time CSRs).  But one can consider these requirements to only apply after mtime is written by firmware (and its value has propagated to all the time CSRs) - and up until firmware writes mtime again.

lixiaogangkkk, 2024-09-27:
> Updating mtime, whenever one does it, is going to be disruptive to the system, especially if jumping backwards in time by more than a very small amount. And when firmware changes or jumps mtime in [...]

Since modifying mtime is suggested to be avoid/careful, why not to move RTC an separate device from mtime. (As most of other architectures (like x86), has set RTC to be separate device from APIC Timer and TSC).

lixiaogangkkk, 2024-09-29:
@gfavor Appreciate the clarification.
If mtime is not RTC, I can't imagine when will writable property be applied, seems it's better to be read-only as most time-stamp-counter.

Thanks,
Steven Li

gfavor, 2024-09-29:
Mtime and time are the time reference for system software.  If system software is using PTP to periodically synchronize system time between systems (as this thread started out being about), then that would imply updating mtime (as well as other relevant "time" devices like an RTC).  Although I don't know if that is what would actually be done (e.g. maybe only an RTC is updated and the mapping between mtime and RTC time is modified accordingly?)

In any case, mtime needs to be writable so that, at a minimum, it can be initialized at system boot and can be re-initialized when exiting a deep system sleep state.

closed 2024-09-29 by Steven-Li-Xiaogang