riscv/riscv-isa-manual #1673: Possibility of 'Environment call from S-mode' delegation to S-mode

evgeniy-paltsev, 2024-10-04:
The common sense about how ECALLs are used gives me impression that S-ECALL shouldn't be delegatable to S-mode.

It was mentioned several times in this repo issues/PRs that S-ECALL shouldn't be delegated to S-mode, i.e.: https://github.com/riscv/riscv-isa-manual/issues/112#issuecomment-341296080, https://github.com/riscv/riscv-isa-manual/pull/712#issuecomment-899973375

In similar case of `Environment call from VS-mode` delegation to VS-mode the docs clearly forbids such delegation - as corresponding bits of `hedeleg` must be `Read-only 0` according to docs.

However, the recent riscv-privileged documentation doesn't have any restrictions about S-ECALL delegation to S-mode, and `scause` register has `9 - Environment call from S-mode` as a valid value. Moreover, the S-ECALL delegation to S-mode still works in spike built from recent sources.

So, does this mean that we allow implementations to have `ECALL from S-mode` delegatable to S-mode?

aswaterman, 2024-10-04:
The spec doesn't mandate that medeleg[9] be hardwired to 0, so it is legal to build an implementation that supports this delegation.  However, setting medeleg[9]=1 would effectively prevent correct implementation of the SBI, and so doing so would in most contexts be considered an M-mode software bug.

aohzju, 2025-02-24:
Actually S mode does not have a way to trigger a CPU exception is not convenient. When porting RTOS, in M mode, we can use ecall to yield CPU for another task. But in S mode, we have to resort to software interrupt, but we have other scenerio which needs to use software interrupt. 
It would be better if each mode has its own environment call instruction, similar to ARM's SMC, HVC and SVC.