riscv/riscv-isa-manual #1674: illegal instruction fault frcsr t0

yf13, 2024-10-06:
Dear experts,

I got exception `Illegal instruction fault(2)` with `mepc=80004d44` and `mtval=0` when running M-mode program on QEMU RiscV target. The code fragment looks like:

```shell
$ riscv64-unknown-elf-objdump -dS sample | fgrep 80004d44: -C1
80004d42: ad6e                  fsd fs11,152(sp)
80004d44: 003022f3            frcsr t0
80004d48: de16                  sw  t0,60(sp)
```

Could someone kindly teach what this exception means here?

aswaterman, 2024-10-06:
There are only two possible explanations: the M-mode software forgot to set mstatus.FS to a nonzero value, or the QEMU instance had floating-point disabled.

allenjbaum, 2024-10-06:
Except epc == 4d44 points to the frcsr, therefore the fsd must have
executed without exception (so mstatus.FS must be nonzero)
 - but if fsd is at 4d42, then frcsr should be at 4d46 - so this is a bit
weird

yf13, 2024-10-07:
many thanks!

aswaterman, 2024-10-07:
The act of loading the float registers in the context-restore code will change FS to Dirty, so you should change FS to Clean after performing the loads.

yf13, 2024-10-07:
> The act of loading the float registers in the context-restore code will change FS to Dirty, so you should change FS to Clean after performing the loads.

Thanks. Then if FS was `Initial` upon context resuming, we shall do `fmv.w.x ft0, zero` alike to set float registers to zero, then set FS to `Clean` afterwards?

aswaterman, 2024-10-07:
That’s the right idea, but in that case, you can set it to Initial rather than Clean.

yf13, 2024-10-08:
thanks again. I'll continue check why FS became 0.

closed 2024-10-08 by yf13