riscv/riscv-isa-manual #1677: LR/SC loop question

yf13, 2024-10-10:
Dear experts,

Can you teach if the following loop is a constrained LR/SC loop:

```assmebly

_lock_loop:
   csrrci a0, mstatus, 8
   csrr   t1, mhartid
   auipc  t0, 0x88
   addi   t0, t0, 1308
   lr.w   t2, (t0)
   bnez   t2, _retry
   li     t2, 1
   sc.w   t2, t2, (t0)
   beqz   _got_lock
_retry:
   csrw   mstatus, a0
   wfi
   j      _lock_loop

_got_lock:

```

Here the LR/SC follows the requirements, but the retry part contains SYSTEM instructions.

allenjbaum, 2024-10-11:
The RISC-V Instruction Set Manual Volume I: Unprivileged Architecture

   -

   The loop comprises only an LR/SC sequence and code to retry the sequence
   in the case of failure, and must comprise at most 16 instructions placed
   sequentially in memory

This is condition is met

   -

   An LR/SC sequence begins with an LR instruction and ends with an SC
   instruction. The dynamic code executed between the LR and SC instructions
   can only contain instructions from the base ''I'' instruction set,
   excluding loads, stores, backward jumps, taken backward branches, JALR,
   FENCE, and SYSTEM instructions. If the ''C'' extension is supported, then
   compressed forms of the aforementioned ''I'' instructions are also
   permitted.

The sequence between LR and SC has no backward jumps or branches or System
ops so it is met.
Note that code could be reordered to move the csrrci and its label to be
just before the lr.w to avoid executing redundant code

   -

   The code to retry a failing LR/SC sequence can contain backwards jumps
   and/or branches to repeat the LR/SC sequence, but otherwise has the same
   constraint as the code between the LR and SC.

This code has a system op (WFI) so fails

   -

   The LR and SC addresses must lie within a memory region with the LR/SC
   eventuality property. The execution environment is responsible for
   communicating which regions have this property.
   -

   The SC must be to the same effective address and of the same data size
   as the latest LR executed by the same hart.

This is met

yf13, 2024-10-11:
@allenjbaum, appreciate your prompt reply.

Are `csrr mstatus,a0` and `csrr t1, mhartid` also `SYSTEM` operations?

allenjbaum, 2024-10-11:
I didn't mention them because I wasn't sure, but just checked and yes, they
are.

aswaterman, 2024-10-12:
I endorse Allen's answer.  The presence of the WFI and CSR instructions in the retry loop are problematic.

If the goal is to save power while the lock is held, you might be able to use the Zawrs extension instead.  Consider:

```
outer_loop:
  lr.w t1, (t0)
  beqz t1, inner_loop
  wrs.nto

inner_loop:
  lr.w t1, (t0)
  bnez t1, outer_loop
  li t2, 1
  sc.w t2, t2, (t0)
  bnez t2, inner_loop
```

yf13, 2024-10-12:
@aswaterman and @allenjbaum, thank you so much!

closed 2024-10-12 by yf13

yf13, 2024-10-12:
Just one more thing, the spec says `backward jump` is not allowed in LRSC sequence, so above line `bnez t1, outer_loop` is not a backward jump, right?

aswaterman, 2024-10-12:
You're right, @yf13.  I rewrote the example to avoid that pitfall.