riscv/riscv-isa-manual #1678: Which interrupt has higher priority between LCOFI and SGEI?

demin-han, 2024-10-11:
section 3.1.9
Multiple  simultaneous  interrupts  destined  for  M-mode  are  handled  in  the  following  decreasing  priority
order: MEI, MSI, MTI, SEI, SSI, STI, LCOFI

section 19.2.3:
Multiple  simultaneous  interrupts  destined  for  HS-mode  are  handled  in  the  following  decreasing  priority
order: SEI, SSI, STI, SGEI, VSEI, VSSI, VSTI.

for LCOFI and SGEI, which has higher priority?

gfavor, 2024-10-12:
For both M-mode and S-mode, LCOFI is lower priority than STI.  So, for consistency, one would expect LCOFI to also be lower priority than VSTI for HS-mode.

@jhauser-us Do you agree?

@demin-han Please go ahead and create a PR that adds LCOFI to the end of the list from section 19.2.3 that you quoted.  @jhauser-us or @aswaterman can then approve the PR (assuming they agree with it).

Thanks.

demin-han, 2024-10-12:
> For both M-mode and S-mode, LCOFI is lower priority than STI. So, for consistency, one would expect LCOFI to also be lower priority than VSTI for HS-mode.
>
> @jhauser-us Do you agree?
>
> @demin-han [...]

In AIA, I found:
[image]

LCOFI has lowest priority in standard interrupt, which is consistent with your said.

aswaterman, 2024-10-12:
Agreed with the above; fix via #1679 has been merged.

closed 2024-10-12 by aswaterman