riscv/riscv-isa-manual #1680: vsm4k.vi is encoded as OPMVV format seems to be imprecise

Steven-Li-Xiaogang, 2024-10-12:
[https://[github.com](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc)/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc](url)
[image]

as vector-coryto section 3.25 described, bit 19~15 is immediate operand, instruction also with ".vi" postfix, so I think using OPIVI format seems to be precise.

Thanks,
Steven-Li

aswaterman, 2024-10-12:
cc @kdockser @nibrunieAtSi5

nibrunieAtSi5, 2024-10-12:
The spec is aligned with riscv-opcodes
https://github.com/riscv/riscv-opcodes/blob/8ab52afb50d25c6798819008252f93576f0d6a43/rv_zvksed#L4
```
vsm4k.vi     31..26=0x21 25=1 vs2 zimm5       14..12=0x2 vd 6..0=0x77
```
`opcode[14:12]` / `funct3` corresponds to `OPMVV`. https://github.com/riscv/riscv-v-spec/blob/2f68ef7256d6ec53e4d2bd7cb12862f406d64e34/v-spec.adoc#101-vector-arithmetic-instruction-encoding

I don't remember what lead to this opcode allocation, @kdockser ?

Looking at the spreadsheet you used for the allocation, https://docs.google.com/spreadsheets/d/128GivlOc7sXiDri5GwJv4ZOlGtq9Lf0HE51YjCPnBvE/edit?usp=sharing, it seems all vector crypto instructions in OP-VE are allocated to `OPMVV`

(Tab: crypto instructions)
[image]

kdockser, 2024-10-17:
Unfortunately, I no longer have access to my notes from that time. However, I seem to recall that the opcode allocation decision was to put the crypto instructions in the "miscellaneous" OPMV[VX] space and reserve the OPIV[VXI] for Integer instructions as these tend to have a need for all three variations.

So, while the OPIVI encoding would be a better match due to its use of an immediate, there was a stronger need to wisely allocate the limited resource of encoding space.

closed 2024-10-17 by kdockser