riscv/riscv-isa-manual #1686: mstatus.VS behavior

jiahzhang, 2024-10-16:
Suppose vstart=N, and a load exception is taken on a vector load corresponding to the element N. Is it ok to leave mstatus.VS as clean in this case? It seems like no observable vector state was changed because of the exception.

If it had trapped on an element after N, then there would be a need to mark mstatus.VS dirty.

aswaterman, 2024-10-17:
If no vector-related state (vcsr, vstart, vl, vtype, V registers) has changed, then it is correct to leave VS as clean.  (It is also OK to change it to Dirty.)

If the trap occurred on a later element, resulting in vstart being changed, then it is required to change VS to dirty.

jiahzhang, 2024-10-17:
That's what I thought thanks

closed 2024-10-17 by jiahzhang