riscv/riscv-isa-manual #1692: Vector Indexed Instructions' order is memory Consistency rule or  just elements' order whose address overlap?

junlingou, 2024-10-21:
There are some vector ordered index instructions, such as ordered indexed load "vloxei8.v", ordered indexed store "vsoxei8.v"; 
as the same time, some unordered index instructions exist, such as unordered indexed load "vluxei8.v", unordered indexed store "vsoxei8.v". what does order meaning?
     Dose order meaning memory Consistency rule?  but there is RVWMO exist, should we make that element is written to memory in elements index order even if the elements' address is not overlap?
    Or the order just meaning we should make the element is written to memory in index order if the the elements' address is not overlap, which will make the final result in memory is the last active element's value for store?

gfavor, 2024-10-21:
Please refer to the "Vector Load/Store Addressing Modes" sub-section in the Vector Loads and Stores section of the RVV chapter in the Unpriv manual.  That sub-section provides a definition of ordered and unordered vector load/store instructions.

AlexGJL, 2024-10-21:
thank you for your answer, the address modeling is clear, but I don't understand the function of ordering

aswaterman, 2024-10-21:
The interesting difference arises when two elements have the same address.  When that happens, it is unspecified which one is executed first for the unordered variants, whereas for the ordered variants, it is required that the element with the lower index is executed first.

Consider an indexed store of a 2-element vector, where the two indices are both zero.  For the unordered case, the contents of memory[rs1] becomes unpredictable: it could contain either vs3[0] or vs3[1].  For the ordered case, it is guaranteed that memory[rs1] will contain vs3[1].

AlexGJL, 2024-10-21:
thanks for your answer  @aswaterman,  I see the order instructions could make sure the final result is the last active element for index store. a. what will happen when two addresses are total different,  Should we make data store to GO (memory ) in index order for order store?   b. as load results are exact values, why we need order load?

aswaterman, 2024-10-21:
(a) Accesses obey RVWMO at the element level, which means they can be reordered if they are going to different addresses.

(b) Because another processor might be storing to those addresses, which makes the reordering become observable.  Suppose another processor stores to an address being accessed by an unordered load.  A load with a lower index might observe the new store, whereas a load with a greater index might not observe the store.  If you use an ordered load, this cannot happen.

That's all the help I can offer for now.

AlexGJL, 2024-10-21:
I understand, thank you @aswaterman.  other non-index order instructions don't  obey RVWMO at the element level, such as Vector Strided Segment Loads and Stores.  making them not obey RVWMO at element level is aim to provide more implement  flexibility ？

aswaterman, 2024-10-21:
All of these relaxations exist to make it easier to build high-performance vector units.

AlexGJL, 2024-10-21:
Ok, Thank you.

closed 2024-10-21 by aswaterman