riscv/riscv-isa-manual #1693: Hypervisor chapter uses term HSLEN but I think it means HSXLEN

james-ball-qualcomm, 2024-10-22:
Quote from chapter below. Should be HSXLEN, not HSLEN I believe.

The hstatus register is an HSXLEN-bit read/write register formatted as shown in [Hypervisor status register (hstatus) when HSLEN=32](https://github.com/riscv/riscv-isa-manual/blob/main/src/hypervisor.adoc#hstatusreg-rv32) when HSXLEN=32 and [Hypervisor status register (hstatus) when HSXLEN=64.](https://github.com/riscv/riscv-isa-manual/blob/main/src/hypervisor.adoc#hstatusreg) when HSXLEN=64. The hstatus register provides facilities analogous to the mstatus register for tracking and controlling the exception behavior of a VS-mode guest.
Hypervisor status register (hstatus) when HSLEN=32

[images/bytefield/hstatusreg-rv32.edn](https://github.com/riscv/riscv-isa-manual/blob/main/src/images/bytefield/hstatusreg-rv32.edn)

gfavor, 2024-10-22:
You're correct re the "HSLEN" typo.  Please go ahead and create a PR to fix that.  Thanks.

james-ball-qualcomm, 2024-10-22:
Cool! My first PR against the new ISA manuals!

kersten1, 2025-01-07:
Closing this one as it appears fixed.

closed 2025-01-07 by kersten1