riscv/riscv-isa-manual #1699: What are the sources of memory attributes under different translation mechanisms?

chara811, 2024-10-28:
1. For single stage address translation (V=0), if PBMT  is not implemented or not enabled, then the memory attributes come from PMA.

2. For two-stage address translation (V=1), if only the G-stage is enabled for address translation, it is equivalent to henvcfg.PBMTE=0. If at this time menvcfg.PBMTE=1, the final memory attributes come from the PBMT of the G-stage leaf PTE (00: PMA, 01: NC, 10: IO). If at this time menvcfg.PBMTE=0, the final memory attributes come from the PMA of the G-stage leaf page.

3. For two-stage address translation (V=1), if only the VS-stage is enabled for address translation, it is equivalent to menvcfg.PBMTE=0, and henvcfg.PBMTE is read-only as 0. The final memory attributes come from the PMA of the VS-stage page. Is there such a thing as address translation with only the VS-stage enabled?

4. For two-stage address translation, is it possible to perform two-stage address translation with menvcfg.PBMTE=0 and henvcfg.PBMTE=0, such as in Sv39x4? Where do the final memory attributes come from then?

Can I understand it this way?

gfavor, 2024-10-28:
Please read the last paragraph of the Svpbmt spec (about two-stage translation).  It specifies how PMAs and PBMT from each stage of translation are combined together.  Note that final memory attributes never simply always come from a stage's PBMT.  Instead there is the concept of selective override of PMAs by PBMT.

chara811, 2024-10-28:
So for two-stage address translation, when G-stage and VS-stage PBMT=0, does the page table attribute of the host OS have higher priority than the PMA of the guest OS?

chara811, 2024-10-28:
For example, if vsatp.mode= sv39, hgatp.mode = bare and menvcfg. PBMTE =0, where does the final memory attribute come from? Is it the PMA of the vs-stage page table?

gfavor, 2024-10-28:
The definition of PBMT=0 is that the PMAs are NOT modified or overriden in any way.

The last paragraph of the spec explicitly defines the priority order in which PBMTs and PMAs are combined together.  If the memory attributes from the PMAs are not overriden by either stage of page tables, then the PMAs determine the final attributes.

closed 2025-01-07 by kersten1